hcomet/CheapoDC #1: Cannot compile firmware

LordWhiteley-UK, 2024-07-05:
I could do with help. I managed to work out to add the ESP32-C3 board and the facility to upload files from a "data" folder - noting that you have to use an older Arduino IDE Iie 1.8.x)
I think I have added all the relevant libraries but when I compile the firmware but I get a load on "undeclared" variables errors. 
eg ledcSetup, ledcAttachPin. Easylogger unqualified-id, Multiple libraries were found for "WiFi.h"

In file included from /Users/johnsilver/Documents/Arduino/CheapoDC/CheapoDC.ino:14:
/Users/johnsilver/Documents/Arduino/libraries/EasyLogger/src/EasyLogger.h:13:18: error: expected unqualified-id before string constant
   13 |     #define endl "\r\n"
      |                  ^~~~~~
/Users/johnsilver/Documents/Arduino/libraries/EasyLogger/src/EasyLogger.h:13:18: error: expected unqualified-id before string constant
   13 |     #define endl "\r\n"
      |                  ^~~~~~
/Users/johnsilver/Documents/Arduino/libraries/EasyLogger/src/EasyLogger.h:13:18: error: expected unqualified-id before string constant
   13 |     #define endl "\r\n"
      |                  ^~~~~~
/Users/johnsilver/Documents/Arduino/CheapoDC/CDController.cpp: In constructor 'dewController::dewController()':
CDController.cpp:20:5: error: 'ledcSetup' was not declared in this scope; did you mean 'CDCSetup'?
   20 |     ledcSetup(CDC_PWM_CHANNEL, CDC_PWM_FREQUENCY, CDC_PWM_RESOLUTION);
      |     ^~~~~~~~~
      |     CDCSetup
CDController.cpp:22:5: error: 'ledcAttachPin' was not declared in this scope; did you mean 'ledcAttach'?
   22 |     ledcAttachPin(CDC_PWM_OUPUT_PIN1, CDC_PWM_CHANNEL);
      |     ^~~~~~~~~~~~~
      |     ledcAttach
Multiple libraries were found for "WiFi.h"
 Used: /Users/johnsilver/Library/Arduino15/packages/esp32/hardware/esp32/3.0.2/libraries/WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
exit status 1
'ledcSetup' was not declared in this scope; did you mean 'CDCSetup'?

hcomet, 2024-07-06:
Yes, I think the Espressif 3.0.x versions started rolling out in may. The build README.md does indicate "use the Espressif version 2.0.x".

I'll admit that I mainly did my development using VSCode with the Arduino plugin. I my check Arduino IDE version was 1.8.19. Should have noted that in the build information

I also note that I did not list ArduinoJSON as a required library. Version 6.x of ArduinoJSON must be used. Version 7.x came out after I published this project. Version 7 has a different memory management model and deprecates dynamicJSON documents which I use.

I just did a fresh install of the 1.8.19 IDE on a new machine and installed a clean set of libraries with appropriate versions. Seems to compile with only the C++17 structured binding warning which is not an issue for the ESP32 compiler.

I'll need to update the build information.

Any other issues?

LordWhiteley-UK, 2024-07-06:
Thankyou for your reply :) yes I realised I needed Arduinojson.h too :)

It’s a blinkin pain that libraries get up-issued but don’t maintain backward compatibility. It took me a while to realise the ledcsetup error was caused because I was using the latest esp32 v3.x in board manager but needed to install v2.x or earlier version :)

Anyway, I managed to compile and upload the firmware and then upload the web interface by uploading the data folder. I’m using the esp32-c3 developer for the board selection with all the board settings left at default 

I can see cheapodc on my router and tried connecting using cheapodc.local. So far so good.
I then get the a login dialogue and enter admin, admin.

However, after this nothing?? In fact re-trying cheapodc.local just throws up an error “can’t find webpage” or something similar.

Checked on router and cheapodc is still showing. So it appears after trying to login, no other webpages from the esp32 can be accessed. I haven’t enabled the web socket API, I don’t really know what’s that’s for.

I don’t if this is because I need to change anything in the Arduino esp32-c3 developer board setup menu?? As I said everything is at default including the esp32 internal memory partition.

It a really neat project which is why I’d really like to get it working - it’s really close to working :)

 PS I’ve designed a 3D printed box on Fusion360 and I’ll stick on Prusa projects  or Thingiverse at some point. 3D printers for Astrophotographers are so useful:)

hcomet, 2024-07-06:
You can try a couple of things:

1 - Monitor it through the serial monitor in the IDE. If you don't see anything then you can increase the level of debug messages by changing the logging level in CDCDefines.h. Change "#define LOG_LEVEL LOG_LEVEL_ERROR" to "#define LOG_LEVEL LOG_LEVEL_INFO" to start.

2 - You can change the weather API service to OpenMeteo so that no API key is needed. Comment out "#define CDC_USE_OPEN_WEATHER" in CDCdefines.h.

3 - confirm that the Blink LED default of 8 will actually work for you. The ESP32-C3 boards seem to be all over the place for pin usage. If pin 8 is used for something else then it may be causing an issue. You try using the blink sketch to see what the right pin is.

hcomet, 2024-07-06:
Okay, I don't have a spare C3 to play with but I did an OTA upload of the image I built using the clean install with clean libraries to one of my completed dew controllers. It seemed to work fine using either Weather API service. I was using a a preconfigured CDCWiFi.json with my network setup. 

Not sure why, but I couldn't find the exact libraries that I seemed to have in my original build environment. It may be a difference between the VSCode arduino library manager and the IDE library manager. Here's the libraries I installed with the Arduino IDE library manager:
[image]

LordWhiteley-UK, 2024-07-06:
Many thanks i will compare with my installed libraries :)
Swapping to OpenMeteo didn't make any difference - I havn't tested using an off board LED (yet) 
Oddly, I get absolutely nothing from the serial monitor ???? Just stays blank which is very odd - I did made the changes to the CDCDefines.h as you suggested, but still no serial monitoring.
It's just strange that I can power up the ESP32 and is clearly connecting to my home network using station mode. And I can get a login dialogue to come up. It just doesn't go the config webpage, after entering admin & admin.
I have definately uploaded the "data" folder, which contains the web pages eg config etc I dont know whether you can remotely check that the data folder exists and that all the webfiles are there. 

FYI - The ESP32C3 I bought is this one from Amazon one (https://www.amazon.co.uk/gp/product/B0D2DH1F1H/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1)

Although experience of trying to get an Onstep system working with a ESP32 R32 - some boards just wont work

hcomet, 2024-07-06:
A lot of people have had issues with the serial monitoring. If you haven't try to enable USB CDC on Boot.

[image]

Also, you could try another ESP32 if you have it. I have compiled and run on both the S3 and classic versions. I only used the C3 for the size and price.

hcomet, 2024-07-08:
Ok, pulled the C3 from one of my dew controllers. I played around and I'm wondering if the issues is the file system. 

I unfortunately did NOT point out in the build instructions to use the data uploader but I also did not mention that I used LittleFS as the file system. NOT SPIFFS. If you installed the older data uploader tool then it only supports SPIFFS. You need the tool found 
[here](https://github.com/lorol/arduino-esp32fs-plugin/releases) - (https://github.com/lorol/arduino-esp32fs-plugin/releases).

This assumes that you modified the CDCdefines.h to set up your WiFi credentials, since uploading using SPIFFS the CDCWiFi.json would not be readable and the CheapoDC would not connect to your router.

If I follow the above path:
- change the CDCdefines.h WiFi info to my AP
- build and upload
- then upload data as SPIFFS

The CheapoDC will connect to my AP, I will be able to log into it BUT it cannot read the filesystem and cannot load the web pages.

As a side note, you need to flip to the serial monitor as soon as the reset via RTS message comes up completing the data upload. Then you should catch the LittleFS filesystem error. If you wait at all then not much gets logged so may look like the serial monitor is not working (Also serial baud rate is 115200 which may be different than your upload baud rate.

In any case hopefully this is the problem you are seeing. Use the tool that supports LittleFS. Upload the data as LittleFS and it should all work. I've been switching back and forth to try things out and switching to LittleFS always fixes things.

LordWhiteley-UK, 2024-07-08:
Many thanks for the continued support :)

I did actually add the eps32fsjar file to tools/ESP32FS/tool folder but I only have one option in the Arduino tools menu  ie ESP32 sketch data upload. When I click on this is seems to upload and then says SPIFFS image uploaded. I don't get the option of using LittleFS '??

I dont get a drop down menu giving option of SPIFFS, LittleFS or FATfs???

I have tried your link and swapped the esp32fs.jar files but it still only uploads SPIFFS & this is the only option on the memory partion ? 

- so I think this is the issue here?. BTW I do have some additional ESP32 modules on the way just in case this generic one is a bummer

hcomet, 2024-07-08:
Yes, if it is only allowing for SPIFFS then that will be your issue. I'm not sure that using another ESP32 module will change whether or not the file system selector shows up. You could try deleting the esp32fs.jar then restart the Arduino IDE. That should remove the tool from the IDE Tools drop down. If it doesn't then the IDE is picking it up from somewhere else. 

I did see that there is a LittleFS data upload tool available for Arduino 2.2.1 or higher. But I have not used the 2.x IDE or this uploader.
https://randomnerdtutorials.com/arduino-ide-2-install-esp32-littlefs/

LordWhiteley-UK, 2024-07-08:
I rolled back from Arduino IDE v2 but maybe I need to re-try this. 
There are other files mentioned that I haven’t installed ie mklittlefs.exe
I’ve been using Ralf S Bacons instructions - https://github.com/RalphBacon/203-SPIFFS-vs-LITTLEFS

LordWhiteley-UK, 2024-07-08:
At last it's working !! I now manged to get the drop-down menu to be able to select littleFS

I removed Arduino IDE and re-installed 
(I retained the libraries under by sketch directory as indicated in Preferences. ie in my case C:\Users\jps\OneDrive\Documents\Arduino\libraries\
In there are the following folders
AduinoJson, AsyncTCP, EasyLogger, ESP32Time, ESPAsyncTCP, ESPAsyncWebSrv & Time

Under C:\Users\jps\OneDrive\Documents\Arduino\tools\ESP32FS\tool is esp32fs.jar
(downloaded from https://github.com/lorol/arduino-esp32fs-plugin/releases)

ESP32 board link that is pasted in Addition Board URL's in Preferences is https://espressif.github.io/arduino-esp32/package_esp32_index.json

hcomet, 2024-07-08:
That's great to hear.

I also tried out the Arduino IDE V2.3.2 with the LittleFS uploader mentioned in the link above. Using the latest 2.0.x Espressif version with libraries above seemed to work fine. A couple had newer subversions which also seem fine as long as you stay within the major version.

I'll work on updating the build documentation and then close this off.

Good luck and thanks for trying out the CheapoDC.

hcomet, 2024-07-15:
Fixed in release v1.0.2

closed 2024-07-15 by hcomet

LordWhiteley-UK, 2024-07-15:
Very Nice Work !!! Thankyou very much for your efforts !!

hcomet, 2024-07-16:
Yes, you are correct the diagram doesn't seem to match the wiring in the completed implementation image.

I'll have to update the diagram.

Thanks again.

hcomet, 2024-07-16:
Ok, diagram updated. Not going to create a new release since there are no code changes and its on the main page for the project in GitHub.

Thanks again.