hcomet/CheapoDC #3: Error opening nonexistent file CDCConfig.json

makahlow, 2025-01-16:
I'm not able to compile a working firmware.

When the program starts, I get

000:00:00:02:018  ALERT     [setup] : Initializing CheapoDC.
000:00:00:02:591  ALERT     [_connectWiFi] : cheapodc.local connected to: Oak with IP address: 192.168.1.92
000:00:00:02:592  ALERT     [_connectWiFi] : mDNS responder started
000:00:00:02:596  ERROR     [LoadConfig] : Failed to open: /CDCConfig.json
000:00:00:02:596  ERROR     [Main-setup] : Load Configuration Failure. Will continue on defaults.

assert failed: tcp_alloc /IDF/components/lwip/lwip/src/core/tcp.c:1851 (Required to lock TCPIP core functionality!)

There is no CDCConfig.json file in the data folder. 

I am able to get littlefs to work in the example files, along with several wifi and networking examples.

I have had to use a memory model with no OTA in order to get it all to fit in memory. The script references a partitions.csv file, but it is not included in the download.

FYI I am using this board - https://www.amazon.com/dp/B0CW62PPZ1?ref=ppx_yo2ov_dt_b_fed_asin_title&th=1. I have attempted to load the software on two separate boards without luck.

hcomet, 2025-01-16:
The CDCConfig.json file is no longer included in the download and is not needed for the first run of the firmware. If the CDCConfig.json is not present then the defaults set in the CDCdefines.h file will be used. These are built into the firmware. As soon as configuration changes are made (via the web interface or API) the CDCConfig.json will be saved to the LittleFS file system.

Your issue seems to be related to the assertion error: 
`assert failed: tcp_alloc /IDF/components/lwip/lwip/src/core/tcp.c:1851 (Required to lock TCPIP core functionality!)`

You didn't mention in your issue which ESP32 Arduino Core version you are using but the assertion error seems to be related to the latest versions of this library, 3.1.0 and 3.1.1 (released in the last month). There are reports of ESPAsyncWebServer failing with the latest ESP32 core. See, [1455](https://github.com/me-no-dev/ESPAsyncWebServer/issues/1455).

CheapoDC makes use of the ESPAsyncWebServer library and there doesn't seem to be a fix available yet. The work around seems to be to back down the version of ESP32 core. Assuming that you have installed either 3.1.0 or 3.1.1 ESP32 core in your Arduino IDE then go back to version 3.0.7 and try rebuilding the CheapoDC. This can be done in the Arduino IDE Boards Manager.

There have also been some reports of 3.1.x increasing build sizes so that may also be your issue with the size of the build. I haven't seen anything related to partitions.csv though. The partitions.csv file is not downloaded though. It is created as part of the build environment based on the ESP32 board you selected in the Arduino IDE and the Partition Scheme.

Try rebuilding with "esp32 by Espressif Systems" version 3.0.7.

hcomet, 2025-01-26:
Note: Arduino core for ESP32 3.1.x now supported with [CheapoDC v2.1.0](https://github.com/hcomet/CheapoDC/releases/tag/v2.1.0)