hcomet/CheapoDC #4: No PWM signal output on pins 0 or 1

makahlow, 2025-01-17:
Thanks again for the help on the issue of compiling

Sorry for the bother, but I've got another issue... The program compiles, connects to my wifi, the web interface works, accesses the weather site, all that. Everything appears to work, except that I'm not getting anything from GPIO0 or 01. Nothing. I've tried subbing other pins (20 & 21) but they don't work either. 

I've built the circuit. I can manually turn on the MOSFETs by applying 3.3V on the trigger/pwm input, and then I get the 12V output as expected.

Everything looks great, it's just that *one little thing* (which I'm hoping is very simple).

hcomet, 2025-01-17:
Maybe a dumb question (comment) but make sure that this line is NOT commented out in CDCdefines.h:
`#define CDC_ENABLE_PWM_OUTPUT // comment out to disable output pins (mainly for debugging)`

If it is commented out then everything will look like its working but no PWM signal is ever sent.

Also, if you haven't tried it yet, switch the Controller Mode to Manual using the Web UI. Then set the Output to something greater than zero. This should force a PWM signal if its working.

Finally, you can try turning on the debug logging, This can be done in the CDCdefines.h.  Change:
`#define LOG_LEVEL LOG_LEVEL_ERROR`
to
`#define LOG_LEVEL LOG_LEVEL_DEBUG`

This will generate A LOT of console logging so you could also filter it down to the functions that set the PWM output by uncommenting and changing:
`// #define LOG_FILTER "handleWebsocketMessage,processor,getCmdProcessor,SaveConfig,processClientRequest"`
to
` #define LOG_FILTER "dewController,updateOutput"`

That should show which pins are set for PWM and when PWM values are changed in the the console log.

Then manually change the output using the Web UI to see of the changes are echoed in the logging.

hcomet, 2025-01-17:
Alright, I found the issue. I had a copy/paste error that I fixed and I obviously messed up the check in. I'll get the fix up tonight.

hcomet, 2025-01-18:
Committed a fix. Grab the new [CDController.cpp](https://github.com/hcomet/CheapoDC/blob/main/CheapoDC/CDController.cpp) and give it a try.

If it all works I'll put out a 2.0.1 release with the fix.

makahlow, 2025-01-18:
Thanks! That appears to work! I'm getting a voltage on the pins which varies with the power applied. Any problems I have taking this to 12V w the MOSFETs is my own problem. ;-)

hcomet, 2025-01-18:
Great to hear. I'll create a release. I did a full retest myself. I use an LED wired for 12V on the MOSFET outputs to make sure it all works since the dew heaters have a bit of lag.

hcomet, 2025-01-18:
Fixed in release [v2.0.1](https://github.com/hcomet/CheapoDC/releases/tag/v2.0.1)

closed 2025-01-18 by hcomet