hcomet/CheapoDC #5: No PWM on GPIO01...

makahlow, 2025-01-18:
My apologies for not checking this out further before saying it was "done".

I only tested the output on pin 1 (GPIO 0). That works. Pin 2 (GPIO 1) does not. When I swap the pin definition in CDCdefines.h to

#define CDC_PWM_OUTPUT_PIN1 1  // set first Output Pin
#define CDC_PWM_OUTPUT_PIN2 0  // set second Output Pin or comment out to disable

(swapping 1 & 2), then it's GPIO 0 that does not work.

hcomet, 2025-01-18:
Another dumb error but I definitely tested both outputs on my device and they both worked. The code definitely shouldn't have worked. It was related to adding support for the latest ESP Arduino Core and my test device is one i upgraded from a previous working release on Core 2.x. I wonder if Pin attachments to PWM channels stay through reboots and power cycles. That would be the only explanation for mine working.

In any case the fix was simple but of course confirming it works would require a fresh chip. Again give the latest version of [CDController.cpp](https://github.com/hcomet/CheapoDC/blob/main/CheapoDC/CDController.cpp) a try.

hcomet, 2025-01-18:
Spent the morning trying to get my device to fail on the second channel. Erased the flash on the C3, tried to detach the channels in code, checked for a short between the pins. The previous code was not setting the second pin properly but it seems that once a pin is attached to a channel it stays there until attached somewhere else. Fix I committed last night should fix it. Added some extra error checking this morning. That's in the latest commit.

Sorry for the issues.

makahlow, 2025-01-18:
Thanks, and no apologies needed. It works now.

On another note - I was having real trouble connecting to wifi, oddly
enough only when the ESP32 was installed in the breadboard. I found a
solution & added the line WiFi.setTxPower(WIFI_POWER_8_5dBm); to the
CDCSetup::_connectWiFi routine, and it cures the problem.

hcomet, 2025-01-19:
Glad to hear its working. 

I have not had any issues with WiFi connectivity but Google indicates that there are versions of the C3 mini boards that have this issue. Happy to add this as a compile time flag in the CDCdefines.h. Something like:
`//#define CDC_ENABLE_WIFI_TX_POWER_MOD WIFI_POWER_8_5dBm // Uncomment to adjust WiFi TX power if you are having connection issues`

Did you add it around line 499 after the connection is established?

Also, related to the first issue you had with the v3.1.x version of Arduino Core. I did some more digging and I may have to move to a different implementation ESPAsyncWebServer that seems to be keeping up with the Espressif changes better. I'll have a look at that later this week.

hcomet, 2025-01-19:
Okay, added/committed the WiFi TX power setting as an option in the CDCdefines.h and put the call right in after the WiFi.begin(...);

Will put out a v2.0.2 release later to close this off.

Thanks.

hcomet, 2025-01-20:
Fixed in v2.0.2 release.

closed 2025-01-20 by hcomet