hcomet/CheapoDC #8: No web access?

LordWhiteley-UK, 2025-01-28:
So, 
(1) Downloaded CDCConfig.json and CDCWiFi.json files from original V1 Cheapodc
(2) CDCConfig.json and CDCWiFi.json files pasted into data folder
(3) Sketch successfully compiled on latest Arduiino IDE and uploaded to my ESP32-C3 without any error.

However, although Cheapodc appears on my network, I cannot open a webpage ie http://cheapodc.

Looking at the Serial Monitor port on the Arduino IDE application, it appears that the ESP32-C3 starts up and generates a whole lot of code BUT then reboots and repeats generating the same code before rebooting again….

hcomet, 2025-01-28:
Okay, a lot of changes between v1.0.3 and the latest v2.1.0. To re-conform it works, I just did a fresh install of v1.0.3 on an ESP32C3 board, configured it and then went through the upgrade steps to take it to v2.1.0. It did work. The first time it booted after the upload of v2.1.0 it did crash and reboot on a LittleFS mount issue. Likely this was caused by the change in partitioning required for v2.1.0. When the LittleFS mount fails then a format of the partition is forced and then a reboot. After this happened I had no issues with the upgraded CheapoDC.

Can you confirm the following:
* make sure you have switched to the new versions of the libraries as outlined in the [Setting Up Your Build Environment](https://github.com/hcomet/CheapoDC/blob/main/CheapoDC/README.md#setting-up-your-build-environment). The old versions don't all work with the latest Arduino core for ESP32.
* In particular make sure you have switched to the new ESPAsyncWebServer and AsyncTCP from Me-No-Dev. The versions from dvarrel do not work with Arduino core for ESP32 version 3.1.x.
* Also, make sure you have the dvarrel versions of the library uninstalled. If the compiler selects the wrong AsyncTCP then it will build but it will crash at runtime.
* Make sure you have the Arduino IDE set properly for the upgrade:
   1. The first time you build and upload make sure that you erase all the flash with: *Erase All Flash Before Sketch Upload: "Enabled"*
   2. Also make sure you've selected the new partition scheme: *Minimal SPIFFS (1.9MB APP with OTA/190KB SPIFFS)*

When the CheapoDC boots the first couple of lines in the Serial Monitor should be similar to:
```
000:00:00:02:017  ALERT     [setup] : Initializing CheapoDC.
000:00:00:02:021  ALERT     [CDCSETUP] : LittleFS Mounted. Size: 131072 Used: 114688
```
Showing that the LittleFS partition mounted and the size. The new partition scheme leaves a very small partition and the data folder with configuration files just fits with about 16K to spare.  Since you said you were getting a lot of output, I assume you're getting past this point so it may be a library version mismatch.

If you're still having problems, upload a copy of the Serial Monitor log with the rebooting.

LordWhiteley-UK, 2025-01-28:
Many Thanks for your reply Stephen :)

It's now working :)
All I had to do was to **_remove_** AsyncTCP by dvarrel from the Arduino IDE
And that fixed it !!
Many Thanks - I must get around to designing a PCB to carry all the electronics :)
Cheers
John

hcomet, 2025-01-28:
No problem John. You're not the first person to have the wrong AsyncTCP in the build path. I'll add an additional note in the upload section asking people to look at the build environment notes.

Glad to hear its working and thanks for using the CheapoDC.

BTW: I have a favour to ask. If you have the time, could you update your review comments for the CheapoDC on the Stellarmate site. The comments currently make it look like you never got it to work.

Thanks,
Stephen.

LordWhiteley-UK, 2025-01-28:
I have asked (emailed) Stellarmate to delete my review, as I cannot change it or add a new review ???

hcomet, 2025-01-28:
Thanks for looking. I had also looked to see if I could add a comment but couldn't.

hcomet, 2025-01-28:
Comment added to upgrade section in readme to check library versions and to check for multiple installed versions of AsyncTCP.

closed 2025-01-28 by hcomet