hcomet/CheapoDC #9: Request in a future iteration

LordWhiteley-UK, 2025-02-03:
Not an issue but a suggestion.
It would be cool to have say 4 MOSFET switches controlled by the ESP32-C3.

You could add say a 0.1ohm in series with each MOSFET and add to the GUI whether switch was on or off and if on, the current draw.

You could add devices eg Camera, mini PC etc and have control over them.

I’m not a coder but the additional circuitry would need 12V divided down for GPIO and each output of the 0.1ohm resistor would also need dividing down.

Or just have a function to turn on or off the 4 output MOSFETs

The PCB I’m designing for the Cheapodc, has 4 DC output that have a 3Amp resettable fuse in series with each, but of course these will always be on.

Anyway it’s just a thought :))

Ps still trying to get stellarmste review removed..

hcomet, 2025-02-03:
I've had similar thoughts about expanding the capabilities of the CheapoDC but more in line with staying with the dew controller focus of the device. Things like adding additional PWM ports or temperature sensor ports etc... Ultimately the issue comes down to what the goal of the project was, a "cheap and easy" dew controller. A basic dew controller that can work for your main telescope and your guide scope which is fairly easy to assemble from low cost "off-the-shelf" parts.

Although good ideas, adding controlled power ports would also be stepping out of the original goal or focus of the project. I think they would also move it more into the area of a power box versus a dew controller. There are projects out there doing power boxes. There was one being built with a lot of discussion on the indilib.org forum about a year ago. I took a quick look but couldn't find a link though.

I can also recommend the projects by Robert Brown. I've built both his myFPESP32 focuser and myDCP4ESP32 dew controller. I wrote the INDI driver for the dew controller. His projects are on [Sourceforge](https://sourceforge.net/u/brownrb/profile/) and his [ESP32DB](https://sourceforge.net/projects/myesp-boards/) project looks like it may be more in line with what you are looking for.

Having said all of that... There may be room for a CheapoPB (Cheapo Power Board) type project separate from this one. I do have a couple of things I'd still like to do for the CheapoDC though. In particular move things like PIN assignment to runtime control from compile time options. I'd like to be able to provide executable firmware installable from the web using ESP32 Web Tools so more runtime flexibility is needed.

Thanks for the ideas. Also, don't worry about the Stellarmate review. You put in your request and now its out of your hands.

LordWhiteley-UK, 2025-02-03:
Yes, I guess I was looking at more of a "Power Box" but your Cheapodc is a unique way of doing a dew controller. 

Yes you could add a thermistor & humidity sensor and not rely on real time internet data but that would be the conventional way of doing it.

It's just with your design, you could simply wire 4 more MOSFET boards to the ESP32, and be able to switch things on/off :) (Current monitoring is not really necessary). The hardest part would be the programming - I cant remember how the Pegasus power box was set up, but I'm sure I used to use one to switch everything off at the end of a NINA imaging session.

I've pretty much designed a PCB for the Cheapodc, I just need to double check the dimensions of the Sourcing Map BNC connectors (As I had to generate the PCB layout by measuring).

I'm also torn with using a linear regulator (LM7805) or a MP1584EN based regulator ? I might go with the MP1584EN as it doesn't need many parts. When I've ordered the PCB's I can rattle off another 3D printed box for it :)

Anyway, the latest version of the Cheapodc was way more easier to setup than the earlier version :)

Many Thanks for all your hard work :)

hcomet, 2025-02-03:
Yes, adding additional control pins is somewhat easy on the hardware side but would require adding control to the web pages, API and probably the INDI driver so they would be usable. Robert Browns ESP32DB is set up this way with runtime configurable pins that can be set to be simple on/off or PWM and whether default state is on or off.

You may want to add a couple of MP1584En based regulators. One to drive the ESP32 and maybe some ports at 5V and one that is adjustable for a variable output. ie: if using a DSLR for a camera they are often voltages  like 7V to 9V.

hcomet, 2025-02-13:
FYI: The commit to the CheapDC-Plus branch seems to be mostly working. You can try it if you'd like by grabbing the [lastest](https://github.com/hcomet/CheapoDC/archive/refs/heads/CheapoDC-Plus.zip) from that branch. 

Only supports the new controller pins via the Web UI. It will take a while to update the INDI driver. I haven't tried the TCP API but it should work. Web Sockets API will definitely take some additional work. I may decide to drop it. Usual upgrade instructions should work.

hcomet, 2025-02-13:
Its just what I called the GitHub branch that I'm making my changes in (typo -  should be CheapoDC-Plus) . Once I'm done I'll merge that branch back into the main branch where I do the releases from. Since the changes will be fairly big and include both code and documentation its easier to work in a separate branch without impacting the main branch. So far I've only made the code changes. Lots of other things to do but the main code seems to be working. You can set up 4 additional pins to be tied to the main Dew Controller pins (Controller mode) or independent and using their own PWM (PWM mode) or just On/Off (Boolean mode). Default is that they are Disabled.

In the Device Management Web UI you'll need to pic a GPIO pin first, Update. Then pick an output mode, Update. Then depending on the mode you can adjust the output.

The link to the zip file will download everything in that branch similar to when you get a zip for the latest release. You can wait for me to finish the full release or you can give it a try now.

It should be backwards compatible with the current driver in INDI/Stellarmate. The two main outputs will work but you won't be able to control the additional outputs via INDI/Stellarmate. Given the release schedule for INDIlib the next release there is likely April so assuming I can get into that release it will be April before a new driver gets picked up.

Thanks for the Stellarmate request.

LordWhiteley-UK, 2025-02-13:
Are you saying then that you could say connect 4 additional MOSFET switches, so that you could have 4 on/off DC ports or PWA or a combination of both??

If my assumption is correct, do you recommend 4 ESP32 pins to use  - there’s 12 GPIO ports on the ESP32 C3 Mini (with the first two controlling the two dew heaters).

Luckily, I haven’t sent off the PCB yet as I want to double check another PCB design I have for a two stepper OnStep controller for the DIY Harmonic Drive EQ mount I designed :)

hcomet, 2025-02-13:
Yes, you could use 4 additional MOSFET switches. You can actually redefine any of the 6 outputs to connect to whatever GPIO pins you want. The software/UI  refers to the pins as Controller Pin 0 through Pin 5. Both Pin 0 and Pin 1 may be disabled or Controller mode pins. You can leave them where they are currently connected (I used GPIO 0 and 1). Pin 2 through 5 can where you want but you which GPIO you pick is dependent on the ESP32-C3 module you are using.

I think you are using the same one I've been using which is the ESP32-C3 Super Mini. Check out the schematic here: [https://michiel.vanderwulp.be/domotica/Modules/ESP32-C3-SuperMini/](https://michiel.vanderwulp.be/domotica/Modules/ESP32-C3-SuperMini/). 

You'll see that GPIO 0, 1, 3, 4, 5, 7, 10, 20, 21 look to be free.  
GPIO2 has a pull up resistor on it.
GPIO6 is set up as an RTC oscillator of some sort.
GPIO8 is the status Led
GPIO9 is the Boot button.

I've tested with GPIOs 0, 1, 3, 4, 5 and 7 so far without issues.

hcomet, 2025-02-13:
Actually not using A2 so not A0 to A5. A2 can be used but it will be pulled-up until software takes over and sets it to zero after boot.

LordWhiteley-UK, 2025-02-13:
Ok noted, so A0 & A1 as before, then A3, A4, A5 and A7 ?

hcomet, 2025-02-13:
You got it.

Good luck,

closed 2025-03-21 by hcomet