hcomet/CheapoDC #12: Outputs not switching with latest firmware 2.2.0??

LordWhiteley-UK, 2025-04-02:
OK, so I have now received the two PCB's I designed. 

One for the original dew controller & and the other one with the 4 switched outputs. Assembled them both and flashed to the latest firmware 2.2.0

The first PCB works perfectly and you can manually turn (ie set to ON and 100%)
on the two dew controllers and so that's all working perfectly.

However, when I connected the second PCB, non of the switched outputs worked.

I initially set Controller Output 2 to GPIO 2 & Update - noticing that you sometimes get a whirling circle on the webpage because something is taking too long to respond??

I then set to Controller mode and ensured on the dew controller page that 100% was set, the dew controllers come
on but non of the outputs and probing GPIO 3,4,5 & 7 are all LOW

NB GPIO2 seems high to be high all the time?

So I'm sure if I've done something wrong or that there is a bug somewhere in the firmware?

[image]
[image]
[image]

hcomet, 2025-04-02:
I don't think GPIO2 was one of the GPIOs that I recommended to use. If you look at the ESP32-C3 SuperMini schematic its tied high with a 10K resistor. You can use it but it will start off high on boot.

[image]

I'm not sure why you're having an issue with the other GPIOs. I'd say to test it with the ESP32-C3 module free of the boards but I see its soldered down. Any chance there's a short pulling them down?

I have used 2.2.0 on several different ESP32-C3 modules without issue.

LordWhiteley-UK, 2025-04-02:
Ok so GPIO is supposed to be high, just something I noticed.
As I said I did monitor GPIO 3,4,5 & 7 and they all remained LOW.
I do have another ESP32 in my second dew controller, so I’ll program that and check the outputs.
However, the mosfet circuits on the second PCB are exactly the same as those on the first PCB which works perfectly.
I did notice that if turned on the dew heaters to 100% and then set controller output 2 to GPIO 3 and controller mode and set 100% - the MOSFET circuit remained off and it turned off the dew outputs?
Anyway, I’ll check the other ESP32 C3 that I have…

hcomet, 2025-04-02:
An idea for checking your board versus the firmware. Try flashing your board with something simpler to test the GPIOs. You could modify the Blink example code to blink all of the GPIOs that you use and then test them to see if they are switching from low to high to low etc...

The Web UI seems to have a random timing issue. It send s off the command for whatever change you input and then commands for updating all the fields in UI. They are all sent asynchronously with a queue waiting for the responses. If a response is missed then it will sit with the waiting spinner running forever. If it seems stuck then just refresh the page. I think the issue is with getting all the responses not with sending the update so a page refresh should fix it.

hcomet, 2025-04-02:
Okay, I have a test unit that I put together which uses 6 dual MOSFETs for 6 outputs. The simplest test using the firmware is to configure them all the same as Controller outputs. 

[image]

Then set the Controller Mode to Manual:

[image]

Then manually modify the Controller Output to change all of the outputs. Here's a video that steps them up from 0% to 100% and back down to 0% by 25% increments. The LEDs are connected to the MOSFET outputs so I know they are working. Outputs 0 and 1 are on the backside. The LED on top is plugged into output 0. 2 through 5 are on the front. If you turn up the sound you can hear me type in the new value and then the mouse click each time I change the output.

https://github.com/user-attachments/assets/fc739b06-1335-468f-812e-05b1f700e6ac

hcomet, 2025-04-03:
That should be doable through the Device Management UI. Just leave the two outputs as Controller mode and then set the other four to Boolean mode. Then the four Boolean mode outputs are independently switchable to either 0% or 100%. Really, they are either On or Off since the outputs are switched to be digital outputs not PWM outputs. 

[image]

If you are using the outputs to control something like a computer then you definitely want them set to Boolean. You do not want a PWM signal going to something dependent on a flat DC output. That could accidentally happen with Controller or PWM mode. Boolean mode can only be fully On or Off. Even if you wrote code to send an API command to set a Boolean mode output to something less than 100% it will only switch fully On.

BTW: the updated INDI driver is also available in the latest version released earlier this week: [https://github.com/indilib/indi/releases/tag/v2.1.3](https://github.com/indilib/indi/releases/tag/v2.1.3) on GitHub. I assumed you used StellarMate but I don't know how long it will take for a StellarMate update to come out. But, then you can also control the four outputs through the StellarMate driver, on the main control tab. This just shows the four outputs can be independently Controller, PWM or Boolean.

[image]

The latest INDI documentation for CheapoDC can be found [here](https://hcomet.github.io/CheapoDC/indi/CheapoDC.html).

LordWhiteley-UK, 2025-04-10:
Yes, it can :)
I've just sent off a new single PCB (components fitted both sides) and I am in the process of designing a 3D enclosure for it. I'll do a write up and post it somewhere???

hcomet, 2025-04-10:
I'm also finishing up a write-up with cases for a 3 output and 6 output build. Both use the off the shelf modules. I'll be posting the write-up on my GitHub Pages which provides web hosting: [https://hcomet.github.io/](https://hcomet.github.io/). If you go to [https://pages.github.com/](https://pages.github.com/) you can set your own up. You'll need to know how to do HTML etc... to post anything though. 

The alternative is to create a repository for it on your GitHub and put everything there. You can do the write-up in markdown which is easier to learn. Your schematics, pcb files and 3d files can all go there. 

In either case, the schematics, pcb files and 3d files should all go in a repository. The write-up goes where you're most comfortable doing it. 

I have know problem referencing it from my site once your done. Just let me know.