hcomet/CheapoDC #14: Temperature and humidity sensor probe

odkmo, 2025-09-03:
I was curious if there was a way to connect a temperature and humidity sensor directly to the board and use these inputs to bypass the need for WiFi?

hcomet, 2025-09-03:
I assume you mean internet more than WiFi. The primary UI and API are only available via WiFi and that cannot be changed.

Currently support for a locally connected temperature and humidity sensor is not supported. That was not the original intent of the the project. But, then again, support for up to 6 outputs was also not part of the original intent. There is development work required to add support for a temperature/humidity sensor to the CheapoDC and unfortunately I'm in the middle of another project. It may take me a couple of months to get to adding this. Also already pushing the boundary of the usable pins on the ESP32-C3 supermini board, so will need to confirm I can physically add a sensor.

If you need a solution now then there are a few other dew controller projects out there. I've built [Robert Brown's ESP32 Dew Controller](https://sourceforge.net/projects/esp32-dew-controller/). It works well although a bit more complicated and involved to build.

Thanks.

hcomet, 2025-11-06:
Support for SHT3x type humidity sensor added in firmware v2.3.0.

closed 2025-11-06 by hcomet